MadLittleMods/node-usb-detection #131: Please provide CPU agnostic binaries of the native module.

egandro, 2021-03-04:
There are some bugs with this:

https://github.com/microsoft/vscode-arduino/issues/1207
https://github.com/MadLittleMods/node-usb-detection/issues/124

The main issue is an architecture specific module:

```
node_modules/usb-detection/build/Release $ file detection.node 

detection.node: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=97745d38cf6d793b9fbccd3dbbc5dae6ad88c705, stripped
```

I don't really know how to handle this. In an enterprise enviornment we would go with a build farm, that is creating binaries for

- Linux (x86_64, x86?, arm64, armv7l)
- macOS (x86_64, arm)
- Windows?

No idea how we should deal with an Opensource project. 

I guess - as Microsoft has an issue with this on VS Code for arm, too - you might get help. @adiazulay is the maintainer of the Arduino Plugin for VS Code.

Maybe we can just ripp the npm in parts? (make linux-x86_64 default and let the community deal with their archs)

- node-usb-detection-driver-linux-x86
- node-usb-detection-driver-linux-arm64
- node-usb-detection-driver-linux-armv7l
- ...

Within the specific driver you provide a getVersion() and this must match the the main implementation?

todbot, 2021-03-04:
This library requires OS-specific bindings compiled for a specific architecture, so I'm not sure what you mean by "CPU agnostic binaries"? As a convenience, the maintainers offer "prebuilds" of the bindings for the most common architectures. It sounds like what you're asking is to provide architecture-specific prebuilds for additional architectures?  I don't understand what you're asking.

MadLittleMods, 2021-03-05:
It's also unclear to me what your question is and what you're asking for.

Maybe you're looking for a rewrite with the N-API and make the code context-aware? There is a open PR for that but still needs some work: https://github.com/MadLittleMods/node-usb-detection/pull/127

egandro, 2021-03-05:
> This library requires OS-specific bindings compiled for a specific architecture, so I'm not sure what you mean by "CPU agnostic binaries"? 

Provide these binaries :) In electron apps - e.g.  like VS the plugins is not recompiled if it is bundled.

MadLittleMods, 2021-03-05:
@egandro There are already [`prebuild`](https://github.com/prebuild/prebuild) binaries for Electron, see https://github.com/MadLittleMods/node-usb-detection/releases/tag/v4.10.0 for example with the whole list.

When installing, if it's not using one of those binaries already for you, which one is missing for your platform?

closed 2021-03-17 by MadLittleMods