MadLittleMods/node-usb-detection #132: This version of Node.js requires NODE_MODULE_VERSION 87.

IvanOnishchenko, 2021-03-17:
If I use Node 13 or 14, node install works well, but when I try to start app (node start) I get error: 
\node_modules\usb-detection\build\Release\detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 79. This version of Node.js requires
NODE_MODULE_VERSION 87. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

But I don't see 87 version on nodej.org, 14 has 83 version, 15 has 88 version.

If I use Node 15, I get error  while node install command running:
$ npm install
npm WARN deprecated xmldom@0.1.31: Deprecated due to CVE-2021-21366 resolved in 0.5.0
npm ERR! code 1
npm ERR! path C:\Users\Администратор\aslice-rekordbox\node_modules\usb-detection
npm ERR! command failed
npm ERR! command C:\Windows\system32\cmd.exe /d /s /c prebuild-install || node-gyp rebuild
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@7.1.2
npm ERR! gyp info using node@15.11.0 | win32 | x64
npm ERR! gyp info find Python using Python version 2.7.18 found at "C:\Python27\python.exe"
npm ERR! gyp info find VS using VS2017 (15.9.28307.1440) found at:
npm ERR! gyp info find VS "C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools"
npm ERR! gyp info find VS run with --verbose for detailed information
npm ERR! gyp info spawn C:\Python27\python.exe
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   'C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\gyp\\gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'msvs',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   'C:\\Users\\Администратор\\aslice-rekordbox\\node_modules\\usb-detection\\build\\config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   'C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   'C:\\Users\\Администратор\\AppData\\Local\\node-gyp\\Cache\\15.11.0\\include\\node\\common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=C:\\Users\\Администратор\\AppData\\Local\\node-gyp\\Cache\\15.11.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=C:\\\\Users\\\\Администратор\\\\AppData\\\\Local\\\\node-gyp\\\\Cache\\\\15.11.0\\\\<(target_arch)\\\\node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=C:\\Users\\Администратор\\aslice-rekordbox\\node_modules\\usb-detection',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'C:\\Users\\Администратор\\aslice-rekordbox\\node_modules\\usb-detection\\build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! Traceback (most recent call last):
npm ERR!   File "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\gyp\gyp_main.py", line 51, in <module>
npm ERR!     sys.exit(gyp.script_main())
npm ERR!   File "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 670, in script_main
npm ERR!     return main(sys.argv[1:])
npm ERR!   File "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 662, in main
npm ERR!     return gyp_main(args)
npm ERR!   File "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 647, in gyp_main
npm ERR!     generator.GenerateOutput(flat_list, targets, data, params)
npm ERR!   File "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\generator\msvs.py", line 2132, in GenerateOutput
npm ERR!     target_list, target_dicts, options, msvs_version
npm ERR!   File "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\generator\msvs.py", line 1936, in _CreateProjectObjects
npm ERR!     guid = _GetGuidOfProject(proj_path, spec)
npm ERR!   File "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\generator\msvs.py", line 985, in _GetGuidOfProject
npm ERR!     guid = guid or MSVSNew.MakeGuid(proj_path)
npm ERR!   File "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\MSVSNew.py", line 51, in MakeGuid
npm ERR!     d = hashlib.md5((str(seed) + str(name)).encode("utf-8")).hexdigest().upper()
npm ERR! UnicodeDecodeError: 'ascii' codec can't decode byte 0xc0 in position 17: ordinal not in range(128)
npm ERR! gyp ERR! configure error
npm ERR! gyp ERR! stack Error: `gyp` failed with exit code: 1
npm ERR! gyp ERR! stack     at ChildProcess.onCpExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\configure.js:351:16)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:378:20)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
npm ERR! gyp ERR! System Windows_NT 10.0.19042
npm ERR! gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd C:\Users\Администратор\aslice-rekordbox\node_modules\usb-detection
npm ERR! gyp ERR! node -v v15.11.0
npm ERR! gyp ERR! node-gyp -v v7.1.2
npm ERR! gyp ERR! not ok

IvanOnishchenko, 2021-03-17:
Also I tried following next steps:

    1. If you have not installed electron-rebuild just install it with the command: `npm i -D electron-rebuild`
    2. Remove from the `node_modules` folder the `usb-detection` folder.
    3. Remove the file `package-lock.json`, this is suggested for ensure a clean installation, but you can try first without remove it.
    4. Run `npm i` to install non-installed modules.
    5. And finally run `./node_modules/.bin/electron-rebuild`

$ ./node_modules/.bin/electron-rebuild
- Searching dependency tree
gyp info find Python using Python version 2.7.18 found at "C:\Python27\python.exe"
gyp info find VS using VS2017 (15.9.28307.1440) found at:
gyp info find VS "C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools"
gyp info find VS run with --verbose for detailed information
gyp info spawn C:\Python27\python.exe
gyp info spawn args [
gyp info spawn args   'C:\\Users\\Администратор\\aslice-rekordbox\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\Администратор\\aslice-rekordbox\\node_modules\\usb-detection\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\Администратор\\aslice-rekordbox\\node_modules\\node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\Администратор\\.electron-gyp\\12.0.1\\include\\node\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\Администратор\\.electron-gyp\\12.0.1',
gyp info spawn args   '-Dnode_gyp_dir=C:\\Users\\Администратор\\aslice-rekordbox\\node_modules\\node-gyp',
gyp info spawn args   '-Dnode_lib_file=C:\\\\Users\\\\Администратор\\\\.electron-gyp\\\\12.0.1\\\\<(target_arch)\\\\node.lib',
gyp info spawn args   '-Dmodule_root_dir=C:\\Users\\Администратор\\aslice-rekordbox\\node_modules\\usb-detection',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'C:\\Users\\Администратор\\aslice-rekordbox\\node_modules\\usb-detection\\build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
Traceback (most recent call last):
  File "C:\Users\▒▒▒▒▒▒▒▒▒▒▒▒▒\aslice-rekordbox\node_modules\node-gyp\gyp\gyp_main.py", line 51, in <module>
    sys.exit(gyp.script_main())
  File "C:\Users\▒▒▒▒▒▒▒▒▒▒▒▒▒\aslice-rekordbox\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 670, in script_main
    return main(sys.argv[1:])
  File "C:\Users\▒▒▒▒▒▒▒▒▒▒▒▒▒\aslice-rekordbox\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 662, in main
    return gyp_main(args)
  File "C:\Users\▒▒▒▒▒▒▒▒▒▒▒▒▒\aslice-rekordbox\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 647, in gyp_main
    generator.GenerateOutput(flat_list, targets, data, params)
  File "C:\Users\▒▒▒▒▒▒▒▒▒▒▒▒▒\aslice-rekordbox\node_modules\node-gyp\gyp\pylib\gyp\generator\msvs.py", line 2132, in GenerateOutput
    target_list, target_dicts, options, msvs_version
  File "C:\Users\▒▒▒▒▒▒▒▒▒▒▒▒▒\aslice-rekordbox\node_modules\node-gyp\gyp\pylib\gyp\generator\msvs.py", line 1936, in _CreateProjectObjects
    guid = _GetGuidOfProject(proj_path, spec)
  File "C:\Users\▒▒▒▒▒▒▒▒▒▒▒▒▒\aslice-rekordbox\node_modules\node-gyp\gyp\pylib\gyp\generator\msvs.py", line 985, in _GetGuidOfProject
    guid = guid or MSVSNew.MakeGuid(proj_path)
  File "C:\Users\▒▒▒▒▒▒▒▒▒▒▒▒▒\aslice-rekordbox\node_modules\node-gyp\gyp\pylib\gyp\MSVSNew.py", line 51, in MakeGuid
    d = hashlib.md5((str(seed) + str(name)).encode("utf-8")).hexdigest().upper()
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc0 in position 17: ordinal not in range(128)
× Rebuild Failed

An unhandled error occurred inside electron-rebuild
node-gyp failed to rebuild 'C:\Users\Администратор\aslice-rekordbox\node_modules\usb-detection'.
Error: `gyp` failed with exit code: 1

Error: node-gyp failed to rebuild 'C:\Users\Администратор\aslice-rekordbox\node_modules\usb-detection'.
Error: `gyp` failed with exit code: 1

    at ModuleRebuilder.rebuildNodeGypModule (C:\Users\Администратор\aslice-rekordbox\node_modules\electron-rebuild\lib\src\module-rebuilder.js:193:19)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)
    at async Rebuilder.rebuildModuleAt (C:\Users\Администратор\aslice-rekordbox\node_modules\electron-rebuild\lib\src\rebuild.js:190:9)
    at async Rebuilder.rebuild (C:\Users\Администратор\aslice-rekordbox\node_modules\electron-rebuild\lib\src\rebuild.js:152:17)
    at async C:\Users\Администратор\aslice-rekordbox\node_modules\electron-rebuild\lib\src\cli.js:146:9

MadLittleMods, 2021-03-17:
Maybe you need to run through these steps: https://stackoverflow.com/a/51027262/796832 (from https://stackoverflow.com/a/51027262/796832)

I just Googled `UnicodeDecodeError: 'ascii' codec can't decode byte 0xc0 in position`. I don't have any insight into building for Electron so Google will probably be a better resource here.

RedskyThirty, 2021-05-10:
@IvanOnishchenko Same issue for me. Have you found a solution?

Symbolk, 2021-05-28:
Same issues+1.

mcous, 2021-10-18:
@IvanOnishchenko it looks like the issue you're describing is a problem with your computer's environment setup for compiling modules with [node-gyp](https://github.com/nodejs/node-gyp) and has nothing to do with usb-detection.

I recommend you try out `usb-detection@4.13.0`, which should have prebuilds available for `electron-rebuild` to download, sidestepping your computer's problems using node-gyp. Additionally, it upgrades to a newer version of `node-gyp` that can use Python 3 instead of Python 2. Given all of Python 2's string encoding problems, that could even be the source of your compilation issues

qlrd, 2022-07-30:
@IvanOnishchenko Issue's and steps reproduced here (with electron 13 and electron-builder).

Resolved only with the solution proposed by @mcous: downgrade the `usb-detection ^4.14.0` to `usb-detection@4.13.0`.