MadLittleMods/node-usb-detection #135: Not working with card readers

agfline, 2021-03-21:
Hi,

It does actually detect my usb card reader when plugging in, but when inserting an SDCARD in the reader nothing happens.
`udevadm monitor -k` detects it though :
```
KERNEL[108063.141070] change   /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/host7/target7:0:0/7:0:0:0/block/sdc (block)
KERNEL[108063.144836] change   /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/host7/target7:0:0/7:0:0:0/block/sdc (block)
KERNEL[108063.144932] add      /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/host7/target7:0:0/7:0:0:0/block/sdc/sdc1 (block)
```

thank you

MadLittleMods, 2021-03-21:
@agfline Looks like you're looking for the same solution that was mentioned in https://github.com/MadLittleMods/node-usb-detection/issues/130

agfline, 2021-03-21:
Thank you for your answer.
I'm already using drivelist. In fact, I'm using usb-detection to detect when a change occurs, then I retrieve disk info using drivelist. 
The problem is that I have a card reader always plugged in, and when I put an SDCARD in it, nothing happens.

todbot, 2021-03-21:
The `node-usb-detection` library just detects USB state changes. The SD card reader is the USB device in this case, and this library detects when it is added or removed.  The SD card is not a USB device, so this library will not detect SD card insert/removal events.  Better to poll `drivelist` occasionally, I guess.

agfline, 2021-03-21:
Yes, but since udev detects that change, and since that change is usb device related (in some way), I hopped it would be possible to get some support for it. Polling would be a bit too cumbersome... I'll give node-udev a try.

Thank you both for your time.

MadLittleMods, 2021-03-22:
I'm not sure of all of the Linux details on how they handle those changes. But to my eye, @todbot summarized it well and the inserted SD card itself is not a USB device. Only the card reader is a USB device.

closed 2021-03-22 by MadLittleMods