MadLittleMods/node-usb-detection #137: using usb-detection with electron

sensslen, 2021-04-04:
Hi there, I am trying to use usb-detection in combination with electron. I do have rebuilt using electron-rebuild but when executing usb-detection functions I get:
```
Worker Thread error Error: Module did not self-register: '\\?\D:\GIT\Private\Cgf.CameraControl.Main.Gui\.webpack\main\native_modules\build\Release\detection.node'.
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1138:18)
    at Module.load (internal/modules/cjs/loader.js:935:32)
    at Function.Module._load (internal/modules/cjs/loader.js:776:14)
    at Module.require (internal/modules/cjs/loader.js:959:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object../node_modules/usb-detection/index.js (D:\GIT\Private\Cgf.CameraControl.Main.Gui\.webpack\main\index.js:124741:18)
    at __webpack_require__ (D:\GIT\Private\Cgf.CameraControl.Main.Gui\.webpack\main\index.js:21:30)
    at Object../node_modules/@sensslen/node-gamepad/NodeGamepad.js (D:\GIT\Private\Cgf.CameraControl.Main.Gui\.webpack\main\index.js:4116:23)
    at __webpack_require__ (D:\GIT\Private\Cgf.CameraControl.Main.Gui\.webpack\main\index.js:21:30)
    at Object../node_modules/@sensslen/node-gamepad/index.js (D:\GIT\Private\Cgf.CameraControl.Main.Gui\.webpack\main\index.js:4384:21)
```

Please feel free to check out the code that produces the issue here: https://github.com/sensslen/Cgf.CameraControl.Main.Gui

Get the issue by going to file->LoadConfiguration loading the following dummy file:
```json
{
  "cams": [
    {
      "type": "PtzLancCamera",
      "instance": 1,
      "ConnectionUrl": "http://1.2.3.4:5001",
      "ConnectionPort": "/dev/ttyACM0"
    }
  ],
  "videoMixers": [
    {
      "type": "blackmagicdesign/Atem",
      "instance": 1,
      "IP": "1.2.3.2",
      "CameraConnections": {
        "1": 1
      }
    }
  ],
  "interfaces": [
    {
      "type": "logitech/F310",
      "instance": 1,
      "VideoMixer": {
        "Connection": 1,
        "MixBlock": 0
      },
      "ConnectionChange": {
        "Default": {
          "up": 1,
          "right": 2,
          "down": 3,
          "left": 4
        }
      },
      "SpecialFunction": {
        "Default": {
        }
      }
    }
  ]
}
```

I am currently stuck. Any help would be highly appreciated!

sensslen, 2021-04-06:
The problem apparently has to do with the way electron loads native modules. This https://github.com/electron/electron/issues/18397 post describes this issue. The Issue I describe has to do with the fact that usb-detection is simply not compatible with electron at the moment.

adkairalla, 2021-04-06:
I did manage to make usb-detection work with electron (had other issues with the way the react flow works) without major incidents. I did a regular
`npm install usb-detection`
inside my package, added it to the package.json as a dependency and had to create a symbolic link in the package.json folder because the `bindings` package was looking for the `.node` file in the wrong place. Also had to run the `electron-rebuild` command for the `usb-detection` package.
With those fixes I managed to make it work inside electron, but I did have issues because I'm not mainly a react programmer and I don't quite understand the concept of the updates and generation of components, so I ended up creating multiple USB listeners each time the page reloaded, hehehe. Hope this helps!