MadLittleMods/node-usb-detection #139: Error installing usb-detection

kimlimjustin, 2021-04-28:
Running classic `npm i usb-detection` get me into an error
```
npm ERR! ../src/detection.cpp: In function ‘void Find(const Nan::FunctionCallbackInfo<v8::Value>&)’:
npm ERR! ../src/detection.cpp:166:50: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
npm ERR!   166 |  uv_queue_work(uv_default_loop(), req, EIO_Find, (uv_after_work_cb)EIO_AfterFind);
npm ERR!       |                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
npm ERR! In file included from ../src/detection.h:5,
npm ERR!                  from ../src/detection.cpp:1:
npm ERR! ../src/detection.cpp: At global scope:
npm ERR! /home/justin/.cache/node-gyp/12.21.0/include/node/node.h:737:7: warning: cast between incompatible function types from ‘void (*)(v8::Local<v8::Object>)’ to ‘node::addon_register_func’ {aka ‘void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)’} [-Wcast-function-type]
npm ERR!   737 |       (node::addon_register_func) (regfunc),                          \
npm ERR!       |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
npm ERR! /home/justin/.cache/node-gyp/12.21.0/include/node/node.h:771:3: note: in expansion of macro ‘NODE_MODULE_X’
npm ERR!   771 |   NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
npm ERR!       |   ^~~~~~~~~~~~~
npm ERR! ../src/detection.cpp:226:1: note: in expansion of macro ‘NODE_MODULE’
npm ERR!   226 | NODE_MODULE(detection, init);
npm ERR!       | ^~~~~~~~~~~
npm ERR! ../src/detection_linux.cpp:1:10: fatal error: libudev.h: No such file or directory
npm ERR!     1 | #include <libudev.h>
npm ERR!       |          ^~~~~~~~~~~
npm ERR! compilation terminated.
npm ERR! make: *** [detection.target.mk:111: Release/obj.target/detection/src/detection_linux.o] Error 1
```

kimlimjustin, 2021-04-28:
Eh, I found a fix for this, running `sudo apt-get install libudev-dev` fill fix this error :)

closed 2021-04-28 by kimlimjustin

MadLittleMods, 2021-04-28:
:+1: Thanks for the follow-up @kimlimjustin!

I wanted to make sure this is documented and I see it in the readme already :+1: https://github.com/MadLittleMods/node-usb-detection#linux