MadLittleMods/node-usb-detection #140: github release url not found when installing

fbede4, 2021-05-05:
I have installed the package via npm, the first thing I realized, that npmjs only has the 4.10.0 release, not the newest 4.11.0.

When installing a got the following error:
    prebuild-install http 404 https://github.com/MadLittleMods/node-usb-detection/releases/download/v4.9.0/usb-detection-v4.9.0-electron-v87-win32-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=12.0.4 runtime=electron arch=x64 libc= platform=win32)

fbede4, 2021-05-05:
Update: with earlier electron versions it seems to work, it only fails with the latest major release (12.x.x)

MadLittleMods, 2021-05-05:
During the 4.11.0 release, I noticed that the prebuilds have been broken since 4.8.0 although there are around 100 prebuilds still available. But 4.11.0 tag had significantly less prebuilds so I decided on holding off publishing to npm, see https://github.com/MadLittleMods/node-usb-detection/pull/134

I still need to spend some more time to look at that PR and make the builds work again. Feel free to pick up the work and submit a PR if you're keen.

MadLittleMods, 2021-10-14:
4.13.0 is now released to npm and [has all of the prebuilds available](https://github.com/MadLittleMods/node-usb-detection/releases/tag/v4.13.0) thanks to @mcous fixing up the CI, https://github.com/MadLittleMods/node-usb-detection/pull/152 🤘

closed 2021-10-14 by MadLittleMods