MadLittleMods/node-usb-detection #145: Failed to install usb-detection in windows-latest environment

kiruh, 2021-07-02:
When I run `npm install` in GitHub Actions `windows-latest` environment I get following error

```
npm ERR! D:\a\cs\cs\electron\node_modules\usb-detection\src\detection.h(5,10): fatal error C1083: Cannot open include file: 'node.h': No such file or directory [D:\a\cs\cs\electron\node_modules\usb-detection\build\detection.vcxproj]
```

Interesting is that each time I run the workflow I get different errors or no errors at all and install succeeds

What's common is that if install fails it does on `node-gyp rebuild` command

`node-gyp rebuild` is called by [usb-detection](https://github.com/MadLittleMods/node-usb-detection/blob/1a6098e2d68e79042327e39a54f17c4f3609c8d9/package.json#L9) and [@serialport/bindings](https://github.com/serialport/node-serialport/blob/ba4e4e8327acaedc6a6bfaf83f335ec5d2a47061/packages/bindings/package.json#L24) packages in their install scripts

It looks like `node-gyp` fails to find some files, e.g. `node.h`, `v8.h`, `common.gypi`

It is really hard to debug it since the outcome alters with each run

Any suggestions on what may be a reason of it?

**Workflow**

```yaml
release_on_windows:
  runs-on: windows-latest

  steps:
    - uses: actions/checkout@master

    - uses: actions/setup-node@master
      with:
        node-version: 14

    - run: npm install
```

**Dependencies**

```json
  "dependencies": {
    "@sentry/electron": "^2.4.1",
    "electron-updater": "^4.3.9",
    "electron-window-state": "^5.0.3",
    "serialport": "^9.0.7",
    "usb-detection": "^4.10.0"
  },
  "devDependencies": {
    "@sentry/cli": "^1.65.0",
    "cross-env": "^7.0.3",
    "electron": "^12.0.9",
    "electron-builder": "^22.9.1",
    "electron-download": "^4.1.1"
  }
```

**Errors**

```
npm ERR! C:\Users\runneradmin\.electron-gyp\12.0.9\include\node\node.h(67,10): fatal error C1083: Cannot open include file: 'v8.h': No such file or directory [D:\a\cs\cs\electron\node_modules\usb-detection\build\detection.vcxproj]
```

```
npm ERR! gyp: C:\Users\runneradmin\AppData\Local\node-gyp\Cache\14.17.1\common.gypi not found (cwd: D:\a\cs\cs\electron\node_modules\usb-detection) while reading includes of binding.gyp while trying to load binding.gyp
```

```
npm ERR! gyp: C:\Users\runneradmin\.electron-gyp\12.0.9\common.gypi not found (cwd: D:\a\cs\cs\electron\node_modules\@serialport\bindings) while reading includes of binding.gyp while trying to load binding.gyp
```

**Log**

```
[... 101 lines cut ...]
    npm ERR! gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\MSBuild\Current\Bin\MSBuild.exe` failed with exit code: 1
    npm ERR! gyp ERR! stack     at ChildProcess.onExit (C:\npm\prefix\node_modules\npm\node_modules\node-gyp\lib\build.js:194:23)
    npm ERR! gyp ERR! stack     at ChildProcess.emit (events.js:375:28)
    npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
    npm ERR! gyp ERR! System Windows_NT 10.0.17763
    npm ERR! gyp ERR! command "C:\\hostedtoolcache\\windows\\node\\14.17.1\\x64\\node.exe" "C:\\npm\\prefix\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
    npm ERR! gyp ERR! cwd D:\a\cs\cs\electron\node_modules\usb-detection
    npm ERR! gyp ERR! node -v v14.17.1
    npm ERR! gyp ERR! node-gyp -v v7.1.2
    npm ERR! gyp ERR! not ok
    
    npm ERR! A complete log of this run can be found in:
    npm ERR!     C:\npm\cache\_logs\2021-06-25T13_32_38_595Z-debug.log
    
                    command='C:\hostedtoolcache\windows\node\14.17.1\x64\node.exe' 'C:\npm\prefix\node_modules\npm\bin\npm-cli.js' rebuild @serialport/bindings@9.0.7 usb-detection@4.10.0
                    workingDir=
npm ERR! code 1
npm ERR! path D:\a\cs\cs\electron
npm ERR! command failed
npm ERR! command C:\Windows\system32\cmd.exe /d /s /c electron-builder install-app-deps

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\npm\cache\_logs\2021-06-25T13_32_38_662Z-debug.log
Error: Process completed with exit code 1.
```

MadLittleMods, 2021-07-06:
Your build shows `VS2019 (16.10.31410.357)`, maybe you need to use an older version of Visual Studio 🤷, https://github.com/MadLittleMods/node-usb-detection#development-compile-from-source

For reference, I am using `VS2017 (15.9.28307.1146)`
```sh
$ npm run node-gyp -- configure --verbose

> usb-detection@4.11.0 node-gyp C:\Users\MLM\Documents\GitHub\node-usb-detection
> node-gyp "configure" "--verbose"

C:\Users\MLM\Documents\GitHub\node-usb-detection {git}{hg}
{lamb} if not defined npm_config_node_gyp (node "C:\Users\MLM\AppData\Roaming\nvm\v14.16.0\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" "configure" "--verbose" )  else (node "C:\Users\MLM\AppData\Roaming\nvm\v14.16.0\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" "configure" "--verbose" )
gyp info it worked if it ends with ok
gyp verb cli [
gyp verb cli   'C:\\Program Files\\nodejs\\node.exe',
gyp verb cli   'C:\\Users\\MLM\\AppData\\Roaming\\nvm\\v14.16.0\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js',
gyp verb cli   'configure',
gyp verb cli   '--verbose'
gyp verb cli ]
gyp info using node-gyp@5.1.0
gyp info using node@14.16.0 | win32 | x64
gyp verb command configure []
gyp verb find Python checking Python explicitly set from command line or npm configuration
gyp verb find Python - "--python=" or "npm config get python" is "C:\Python27"
gyp verb find Python - executing "C:\Python27" to get executable path
gyp verb find Python - "C:\Python27" is not in PATH or produced an error
gyp verb find Python Python is not set from environment variable PYTHON
gyp verb find Python checking if "python" can be used
gyp verb find Python - executing "python" to get executable path
gyp verb find Python - executable path is "C:\Users\MLM\AppData\Local\Programs\Python\Python37-64\python.exe"
gyp verb find Python - executing "C:\Users\MLM\AppData\Local\Programs\Python\Python37-64\python.exe" to get version
gyp verb find Python - version is "3.7.7"
gyp info find Python using Python version 3.7.7 found at "C:\Users\MLM\AppData\Local\Programs\Python\Python37-64\python.exe"
gyp verb get node dir no --target version specified, falling back to host node version: 14.16.0
gyp verb command install [ '14.16.0' ]
gyp verb install input version string "14.16.0"
gyp verb install installing version: 14.16.0
gyp verb install --ensure was passed, so won't reinstall if already installed
gyp verb install version is already installed, need to check "installVersion"
gyp verb got "installVersion" 9
gyp verb needs "installVersion" 9
gyp verb install version is good
gyp verb get node dir target node version installed: 14.16.0
gyp verb build dir attempting to create "build" dir: C:\Users\MLM\Documents\GitHub\node-usb-detection\build
gyp verb build dir "build" dir needed to be created? null
gyp verb find VS msvs_version was set from command line or npm config
gyp verb find VS - looking for Visual Studio version 2017
gyp verb find VS VCINSTALLDIR not set, not running in VS Command Prompt
gyp verb find VS checking VS2017 (15.9.28307.1146) found at:
gyp verb find VS "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community"
gyp verb find VS - found "Visual Studio C++ core features"
gyp verb find VS - found VC++ toolset: v141
gyp verb find VS - found Windows SDK: 10.0.17763.0
gyp info find VS using VS2017 (15.9.28307.1146) found at:
gyp info find VS "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community"
gyp info find VS run with --verbose for detailed information
gyp verb build/config.gypi creating config file
gyp verb build/config.gypi writing out config file: C:\Users\MLM\Documents\GitHub\node-usb-detection\build\config.gypi
gyp verb config.gypi checking for gypi file: C:\Users\MLM\Documents\GitHub\node-usb-detection\config.gypi
gyp verb common.gypi checking for gypi file: C:\Users\MLM\Documents\GitHub\node-usb-detection\common.gypi
gyp verb gyp gyp format was not specified; forcing "msvs"
gyp info spawn C:\Users\MLM\AppData\Local\Programs\Python\Python37-64\python.exe
gyp info spawn args [
gyp info spawn args   'C:\\Users\\MLM\\AppData\\Roaming\\nvm\\v14.16.0\\node_modules\\npm\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\MLM\\Documents\\GitHub\\node-usb-detection\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\MLM\\AppData\\Roaming\\nvm\\v14.16.0\\node_modules\\npm\\node_modules\\node-gyp\\addon.gypi',
[... 11 lines cut ...]
gyp info spawn args   '--generator-output',
gyp info spawn args   'C:\\Users\\MLM\\Documents\\GitHub\\node-usb-detection\\build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info ok
```

The build completes succesfully:
```sh
$ npm run rebuild

> usb-detection@4.11.0 rebuild C:\Users\MLM\Documents\GitHub\node-usb-detection
> node-gyp rebuild

C:\Users\MLM\Documents\GitHub\node-usb-detection {git}{hg}
{lamb} if not defined npm_config_node_gyp (node "C:\Users\MLM\AppData\Roaming\nvm\v14.16.0\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Users\MLM\AppData\Roaming\nvm\v14.16.0\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  detection.cpp
  deviceList.cpp
  detection_win.cpp
  win_delay_load_hook.cc
     Creating library C:\Users\MLM\Documents\GitHub\node-usb-detection\build\Release\detection.lib and object C:\Users\MLM\Documents\GitHub\node-usb-detection\build\
  Release\detection.exp
  detection.vcxproj -> C:\Users\MLM\Documents\GitHub\node-usb-detection\build\Release\\detection.node
```

kiruh, 2021-07-07:
@MadLittleMods I tried to use different versions of Visual Studio but the version doesn't seem to be an issue.

It's possible to rebuild it with [VS2019](https://github.com/actions/virtual-environments/blob/main/images/win/Windows2019-Readme.md#visual-studio-enterprise-2019).

**Verbose output of successful run with VS2019:**

```
Run npm install
  npm install
  shell: C:\Program Files\PowerShell\7\pwsh.EXE -command ". '{0}'"
  env:
    DEBUG: electron-builder

> swimcloud-meet@0.1.0 postinstall
> electron-builder install-app-deps

  • electron-builder  version=22.10.5
  • loaded configuration  file=package.json ("build" field)
  • spawning        command=D:\a\cs\cs\electron\node_modules\app-builder-bin\win\x64\app-builder.exe node-dep-tree --dir D:\a\cs\cs\electron
  • unresolved deps  unresolved=blend-of-streamfs-constantsinherits nodeModuleDir=D:\a\cs\cs\electron\node_modules round=0
  • unresolved deps  unresolved=inheritsstring_decoderutil-deprecate nodeModuleDir=D:\a\cs\cs\electron\node_modules round=0
  • unresolved deps  unresolved=bufferinherits nodeModuleDir=D:\a\cs\cs\electron\node_modules round=0
  • unresolved deps  unresolved=inheritsstring_decoderutil-deprecate nodeModuleDir=D:\a\cs\cs\electron\node_modules round=0
  • unresolved deps  unresolved=bindingseventemitter2nan nodeModuleDir=D:\a\cs\cs\electron\node_modules round=0
  • unresolved deps  unresolved=mkdirp-classicnapi-build-utilsnpmlogrcexpand-templateminimistnoop-loggerpumpsimple-getwhich-pm-runsdetect-libcnode-abitunnel-agentgithub-from-packagetar-fs
                     nodeModuleDir=D:\a\cs\cs\electron\node_modules
                     round=0
  • exited          command=app-builder.exe code=0 pid=140 out=[{"dir":"D:\\a\\cs\\cs\\electron\\node_modules","deps":[{"name":"@serialport/binding-abstract","version":"9.0.7"},{"name":"@serialport/binding-mock","version":"9.0.7"},{"name":"@serialport/bindings","version":"9.0.7","hasPrebuildInstall":true},{"name":"@serialport/parser-byte-length","version":"9.0.7"},{"name":"@serialport/parser-cctalk","version":"9.0.7"},{"name":"@serialport/parser-delimiter","version":"9.0.7"},{"name":"@serialport/parser-inter-byte-timeout","version":"9.0.7"},{"name":"@serialport/parser-readline","version":"9.0.7"},{"name":"@serialport/parser-ready","version":"9.0.7"},{"name":"@serialport/parser-regex","version":"9.0.7"},{"name":"@serialport/stream","version":"9.0.7"},{"name":"ansi-regex","version":"2.1.1"},{"name":"aproba","version":"1.2.0"},{"name":"are-we-there-yet","version":"1.1.5"},{"name":"base64-js","version":"1.5.1"},{"name":"bindings","version":"1.5.0"},{"name":"bl","version":"4.1.0"},{"name":"buffer","version":"5.7.1"},{"name":"chownr","version":"1.1.4"},{"name":"code-point-at","version":"1.1.0"},{"name":"console-control-strings","version":"1.1.0"},{"name":"core-util-is","version":"1.0.2"},{"name":"debug","version":"4.3.1"},{"name":"decompress-response","version":"4.2.1"},{"name":"deep-extend","version":"0.6.0"},{"name":"delegates","version":"1.0.0"},{"name":"detect-libc","version":"1.0.3"},{"name":"electron-window-state","version":"5.0.3"},{"name":"end-of-stream","version":"1.4.4"},{"name":"eventemitter2","version":"5.0.1"},{"name":"expand-template","version":"2.0.3"},{"name":"file-uri-to-path","version":"1.0.0"},{"name":"fs-constants","version":"1.0.0"},{"name":"gauge","version":"2.7.4"},{"name":"github-from-package","version":"0.0.0"},{"name":"graceful-fs","version":"4.2.6","optional":true},{"name":"has-unicode","version":"2.0.1"},{"name":"ieee754","version":"1.2.1"},{"name":"inherits","version":"2.0.4"},{"name":"ini","version":"1.3.8"},{"name":"is-fullwidth-code-point","version":"1.0.0"},{"name":"isarray","version":"1.0.0"},{"name":"jsonfile","version":"4.0.0"},{"name":"mimic-response","version":"2.1.0"},{"name":"minimist","version":"1.2.5"},{"name":"mkdirp","version":"0.5.5"},{"name":"mkdirp-classic","version":"0.5.3"},{"name":"ms","version":"2.1.2"},{"name":"nan","version":"2.14.2"},{"name":"napi-build-utils","version":"1.0.2"},{"name":"node-abi","version":"2.26.0"},{"name":"noop-logger","version":"0.1.1"},{"name":"npmlog","version":"4.1.2"},{"name":"number-is-nan","version":"1.0.1"},{"name":"object-assign","version":"4.1.1"},{"name":"once","version":"1.4.0"},{"name":"prebuild-install","version":"6.1.2"},{"name":"process-nextick-args","version":"2.0.1"},{"name":"pump","version":"3.0.0"},{"name":"rc","version":"1.2.8"},{"name":"readable-stream","version":"2.3.7"},{"name":"safe-buffer","version":"5.1.2"},{"name":"semver","version":"5.7.1"},{"name":"serialport","version":"9.0.7"},{"name":"set-blocking","version":"2.0.0"},{"name":"signal-exit","version":"3.0.3"},{"name":"simple-concat","version":"1.0.1"},{"name":"simple-get","version":"3.1.0"},{"name":"string-width","version":"1.0.2"},{"name":"string_decoder","version":"1.1.1"},{"name":"strip-ansi","version":"3.0.1"},{"name":"strip-json-comments","version":"2.0.1"},{"name":"tar-fs","version":"2.1.1"},{"name":"tar-stream","version":"2.2.0"},{"name":"tunnel-agent","version":"0.6.0"},{"name":"usb-detection","version":"4.10.0","hasPrebuildInstall":true},{"name":"util-deprecate","version":"1.0.2"},{"name":"which-pm-runs","version":"1.0.0"},{"name":"wide-align","version":"1.1.3"},{"name":"wrappy","version":"1.0.2"}]},{"dir":"D:\\a\\cs\\cs\\electron\\node_modules\\bl\\node_modules","deps":[{"name":"readable-stream","version":"3.6.0"}]},{"dir":"D:\\a\\cs\\cs\\electron\\node_modules\\tar-stream\\node_modules","deps":[{"name":"readable-stream","ver [... cut ...]
    npm timing build:run:install:node_modules/@serialport/bindings Completed in 7136ms
    npm timing build:run:install Completed in 7136ms
    npm timing build:deps Completed in 7137ms
    npm timing build Completed in 7137ms
    npm timing command:rebuild Completed in 7347ms
    npm verb exit 0
    npm timing npm Completed in 7588ms
    npm info ok 
    
  • exited          command=app-builder.exe code=0 pid=4732

added 365 packages, and audited 366 packages in 2m

found 0 vulnerabilities
```

More interesting is why it is not idempotent...

timebuzzerfelix, 2021-09-19:
same problem here. If I run yarn, it works 2 of 10 times, without changing anything, just run again. 

I tried rebuild with a step in case of an error:

  if: failure()
        run: |
          yarn electron-rebuild -f -w usb-detection
          yarn

didn' help. Any idea how to fix this?

kiruh, 2021-09-19:
No, I just run it a couple of times and hope it works 🙂

timebuzzerfelix, 2021-09-19:
I had the same problem on a local build server, but only on the first run. After that it worked. 
I guess something was "not ready". On GitHub Actions every time a new clean machine is created, maybe thats the reason why it fals that often

kiruh, 2021-09-19:
I suppose initial state of GitHub Action machines differs between runs 🧐

I understand that the second run is always successful, right? If so I'll
also use retry on fail 😁

timebuzzerfelix, 2021-09-19:
only on a local machine where I installed python etc. manually (and keep it installed)

Do you know a way how to tell GitHub Actions to rerun after fail let's say for 10 times? would increase the change of successful automated building :D

kiruh, 2021-09-20:
Not sure if it's possible. I think you still can define a task twice and
execute second time only if first one failed. But it feels like a duct tape
😅

closed 2021-10-18 by kiruh