MadLittleMods/node-usb-detection #150: Failed test on Raspberry Pi 400

yiancar, 2021-08-10:
After compiling the module for armv7l, I run the test which failed on device disconnect:

```
Randomized with seed 46328
Started
..F.Add/Insert or Remove a USB device
.Add/Insert a USB device
.Remove a USB device
.DEPRECATION: An asynchronous before/it/after function was defined with the async keyword but also took a done callback. This is not supported and will stop working in the future. Either remove the done callback (recommended) or remove the async keyword. (in spec: usb-detection API `.find` should return a promise when vid is given)
.DEPRECATION: An asynchronous before/it/after function was defined with the async keyword but also took a done callback. This is not supported and will stop working in the future. Either remove the done callback (recommended) or remove the async keyword. (in spec: usb-detection API `.find` should return a promise when vid and pid are given)
...

Failures:
1) usb-detection can exit gracefully after `startMonitoring` then an async delayed `stopMonitoring`
  Message:
    Failed
  Stack:
    Error: Failed
        at <Jasmine>
        at /home/pi/Documents/keyboard_development/via-config-private/node_modules/usb-detection/test/test.js:157:11

11 specs, 1 failure
Finished in 16.992 seconds
Randomized with seed 46328 (jasmine --random=true --seed=46328)
Segmentation fault
```

MadLittleMods, 2021-08-10:
To clarify, did you actually plug and unplug USB devices as it was going (the tests require some manual interaction)? It looks like maybe given the rest of them passed.

`armv7l` isn't a platform I am going to be able to test and support myself. Feel free to send a PR if you want to dig into it.

yiancar, 2021-08-10:
I have indeed yes.
I have also tried using the 2 suggested PRs to stop it from segfaulting on Linux. That part has worked.

I will try my best to try and figure it out

yiancar, 2021-08-10:
Interestingly, if you are able to give any pointers, I have noticed the following:

Tests that start with remove a USB device, fail if I first insert a device before removing it. Is this what it should happen? Or should the module be able to detect the new device and then detect that it has been removed?

MadLittleMods, 2021-08-10:
@yiancar I think it should work. I haven't had any issues when testing on Windows or macOS when I just plug and unplug the same USB device over and over.

yiancar, 2021-08-16:
This was all incorrect, this module works fine on raspian :)

closed 2021-08-16 by yiancar

MadLittleMods, 2021-08-17:
@yiancar Can you explain how you came to this realization? What's different from when you originally posted to now?

yiancar, 2021-08-17:
Of course.
In my case usb-detection is a module used under a program I was debugging. In general this program is suppose to execute some code when a device is plugged/unplugged. This was not happening on the Pi the majority of the times.

After running the test suite, some of the test were failing as seen above. The tests expect that a detection will happen within 10 seconds. It seems the Pi can be a bit slow and sometimes it takes around 10 seconds for the detection to happen, adding that up with the initial time it takes to plug/unplug a device and we have a false failure.

Increasing the test timeout time lets all the tests pass on a Pi. Also manually creating a program that print detentions showed that all devices, sometimes after a bit of a delay, get detected.

Cheers