MadLittleMods/node-usb-detection #151: 4.11 tag not published to npm

mcous, 2021-08-12:
The [v4.11.0 release](https://github.com/MadLittleMods/node-usb-detection/releases/tag/v4.11.0) is [not currently published to npm](https://www.npmjs.com/package/usb-detection), where the latest is 4.10.0.

Not sure what the publishing strategy for this module is, but if the lack of publishing is related to [the CI issues](https://github.com/MadLittleMods/node-usb-detection/pull/134) you've been seeing, I'm happy to help out and try to PR a fix if you have any pointers!

MadLittleMods, 2021-08-12:
@mcous The CI build issues is what's stopping them from being published to npm. The prebuilds aren't available for all platforms so I haven't pushed it to npm yet, see https://github.com/MadLittleMods/node-usb-detection/issues/140#issuecomment-832858664

I'm not sure what's causing the CI failure, you can see the [failing job](https://travis-ci.org/github/MadLittleMods/node-usb-detection/jobs/761538461) linked from the [PR](https://github.com/MadLittleMods/node-usb-detection/pull/134). I was just trying to clear out all of the old stuff and update the dependencies there but that didn't seem to change things.

MadLittleMods, 2021-10-14:
4.13.0 is now published to npm and [has prebuilds](https://github.com/MadLittleMods/node-usb-detection/releases/tag/v4.13.0) thanks to the contribution from @mcous, https://github.com/MadLittleMods/node-usb-detection/pull/152 🤘

closed 2021-10-14 by MadLittleMods