MadLittleMods/node-usb-detection #154: Move prebuild system to `prebuildify` and `node-gyp-build`

mcous, 2021-10-20:
The [prebuild-install](https://github.com/prebuild/prebuild-install) dependency now advocates against its own use:

> **Instead of [`prebuild`](https://github.com/prebuild/prebuild) paired with [`prebuild-install`](https://github.com/prebuild/prebuild-install), we recommend [`prebuildify`](https://github.com/prebuild/prebuildify) paired with [`node-gyp-build`](https://github.com/prebuild/node-gyp-build).**

The benefits touted by `prebuildify` and `node-gyp-build` _seem_ like useful upgrades to the build system that could meaningfully reduce the number of issues faced by Electron users of this library (e.g. #153). I'm not sure, but it looks like there's a _chance_ it could make it easier / possible to start shipping a universal macOS prebuilt binary to resolve #141.

> With prebuildify, all prebuilt binaries are shipped inside the package that is published to npm, which means there's no need for a separate download step like you find in prebuild. The irony of this approach is that it is faster to download all prebuilt binaries for every platform when they are bundled than it is to download a single prebuilt binary as an install script.

(Aside: I'm about to go on several weeks of vacation, and I intend to look into this when I'm back. Until that time, though, if someone else picked this ticket up, I would not be disappointed!)

thegecko, 2021-11-23:
We recently ported [node-usb](https://github.com/node-usb/node-usb) to `prebuildify` using `prebuildify-cross`, it may serve as inspiration.

Julusian, 2021-11-23:
I would be concerned about how big a nan based package would end up being. It produces a rather long list of targets, which will only get worse if adding prebuildify-cross to the mix

I really need to finish up that node-api rewrite I started...