MadLittleMods/node-usb-detection #158: Segmentation fault when removing and adding the usb too fast

DHVPP, 2022-02-11:
Hello,

I've encountered a problem with this package where If i remove and add the USB device too fast (as fast as I can), my Electron app just crashes and I get back the `Segmentation fault (core dumped)` error in logs. This error specifically happens when the package takes more than 1 sec to detect changes on USB ports (this happens kinda often) and in the meantime i pull out and plug in the USB again.

Info about the implementation: I call `usbDetect.startMonitoring()` as soon as the app is ready to launch, and I call `usbDetect.stopMonitoring()` before the app closes. I'm also only listening to `add` and `remove` events. 

Linux Ubuntu `20.04`
Node version `16.13.1`
Electron version `16.0.0`

I've seen #138 and #156 issues, but this might be a different problem.

Thanks for reading, hope to hear from you about this,
Regards

DHVPP, 2022-02-17:
@MadLittleMods Hey, sorry to be bothering you directly. Can I at least get a feedback if this sounds to you like a valid case or it might be a problem with my project specificaly?

MadLittleMods, 2022-02-17:
@DHVPP Sounds plausible which is why the labels were added 👍

oleszczyk, 2022-10-27:
Same issue here :(