MadLittleMods/node-usb-detection #159: symbol lookup error

nishad777, 2022-02-15:
I got this error when i try to start my electron app. electron version 14. usb-detection 4.13.0
/node_modules/electron/dist/electron remote.js: symbol lookup error: /node_modules/usb-detection/build/Release/detection.node: undefined symbol: _ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEiNS_19ConstructorBehaviorENS_14SideEffectTypeEPKNS_9CFunctionE

IonutBurlacu, 2022-03-20:
I'm getting a similar error as the one above:
```dyld: lazy symbol binding failed: Symbol not found: __ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEiNS_19ConstructorBehaviorENS_14SideEffectTypeEPKNS_9CFunctionEhhh
  Referenced from: /Users/burlacuionutmihai/Desktop/mokiapp-electron/app/node_modules/usb-detection/build/Release/detection.node
  Expected in: flat namespace

dyld: Symbol not found: __ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEiNS_19ConstructorBehaviorENS_14SideEffectTypeEPKNS_9CFunctionEhhh
  Referenced from: /Users/burlacuionutmihai/Desktop/mokiapp-electron/app/node_modules/usb-detection/build/Release/detection.node
  Expected in: flat namespace```

brenthompson, 2022-05-02:
Got the same error as @IonutBurlacu on electron 13 & usb-detection 14.4.1. `electron-rebuild` didn't appear to do anything, it returned immediately. although it did replace the .node file. I went into node_modules/usb-detection and ran a `yarn install` and that at least resolved the electron crash on startup due to missing symbols, but usb-detection still wouldn't work, got an exception when trying to `require` it. I added a postinstall script to my package.json to run `electron-builder install-app-deps` and that fixed everything up