MadLittleMods/node-usb-detection #160: NODE_MODULE_VERSION 93. This version of Node.js requires NODE_MODULE_VERSION 83. Very good, I need urgent help, this error keeps coming up constantly and I can't find any solution, thanks for your help

Diegoecoms, 2022-02-15:
was compiled against a different Node.js version using
NODE_MODULE_VERSION 93. This version of Node.js requires
NODE_MODULE_VERSION 83. Please try re-compiling or re-installing
the module (for instance, using npm rebuild or npm install).

todbot, 2022-02-15:
Sounds like you need to run `electron-rebuild`

MadLittleMods, 2022-02-16:
It sounds like `prebuild` is choosing the wrong binary for you. There are v83 binaries available so I'm not sure why it would choose v93, https://github.com/MadLittleMods/node-usb-detection/releases/tag/v4.13.0

 1. What OS are you running on?
 1. What do you see when you run `node --version`?
 1. Delete `node_modules/usb-detection`, then run `npm install usb-detection`, what is the full console output?

Diegoecoms, 2022-02-16:
I'm a total beginner, first of all thanks for the help, sorry I hardly understand anything about programming, I'm on windows 10, what exactly do you mean by What do you see when you run node --version? And what do you mean by full console output? sorry for the bother and thank you very much for the help

MadLittleMods, 2022-02-16:
How are you encountering this error? Can you share a screenshot so I have more context on what you're doing?

The console/terminal is what you type commands in. I assume you're making a Node.js application and running the app.

Diegoecoms, 2022-02-16:
[image]

Diegoecoms, 2022-02-16:
Thank you very much for the help, really.

closed 2022-02-16 by MadLittleMods

c0rrell, 2022-10-19:
Having the same issue as you, @MadLittleMods 

Did you find a way to fix it? 🤔