MadLittleMods/node-usb-detection #170: Localization for ARM64

eyq93811, 2022-09-05:
hello author， Feedback on the problem of reporting errors in the installation of dependency packages for domestic adaptation of the ARM64 architecture. Is there a solution?

MadLittleMods, 2022-09-05:
@eyq93811 I have no idea what you're saying. Please share the actual error you're running into.

eyq93811, 2022-09-06:
> @eyq93811 I have no idea what you're saying. Please share the actual error you're running into.

Hello author, I am using the usb-detection installation package on a Linux general machine to report an error, as follows
[image]

MadLittleMods, 2022-09-06:
I still feel like I am missing what the problem is but I guess it's a problem using this in Electron on an ARM64 platform

eyq93811, 2022-09-07:
> 我仍然觉得我错过了问题所在，但我想在 ARM64 平台上的 Electron 中使用它是一个问题

好的，谢谢回复