MadLittleMods/node-usb-detection #174: balena docker container not working

iKK001, 2022-12-26:
Any idea why this library does not work when executed inside a balenaOS docker container ?

I am not sure whether it is the library or whether it is the combination within the docker service.

Fact is that `lsusb` cmd works when logged into the running docker image. Also `/dev` shows the USB-device correctly as `ttyUSB0`.

Everything works when locally running my code on a Mac. I can detect plugging a USB-device and the `usb-detection` library gives me the callback:

```
import usbDetect from "usb-detection"
usbDetect.startMonitoring()
usbDetect.on("add", () => {
     // only works when run locally on a Mac - but does not work inside a bridged balenaOS docker service !!!!!!!
}
```

I do `ENV UDEV=1` in my Dockerfile and add `priviledge: true` to the service in docker-compose.yml file. But it does not seem to work.

Any idea why ??

Julusian, 2022-12-28:
I have a feeling that on mac docker containers are run inside a linux vm. Or there will be some other translation layer going on as docker is a linux only thing.
So maybe it will work if you can get the usb devices passed through into that docker vm/whatever, but in doing so they may no longer be available to macos.

A while back I tried to make a node application which used usb to work in docker (only considering being run on a linux machine), and gave up trying to get that working as there were many problems to workaround with doing usb in docker

MadLittleMods, 2022-12-28:
@iKK001 Please update this issue with your workaround and cross-link your forum post: https://github.com/MadLittleMods/node-usb-detection/issues/123#issuecomment-1365276924